bit-bots/bitbots_vision #18: Use ROS internal loggers instead of debug printer

timonegk, 2019-05-14:
Currently, the vision debug printer just reimplements the actual ROS named loggers. You can provide a logger name like this: `rospy.logdebug('message', logger_name='field_boundary_detector')`. All logged messages will be shown and can be filtered int the rqt log console and the logger leven in the actual screen output can be controlled by using the rqt logger level plugin or by calling the `set_logger_level` service. The logger name in the above example would be `/rosout/<node name>/field_boundary_detector`.
Using these loggers would also reduce external dependencies of the modules.

NFiedler, 2019-05-14:
I strongly support this! This should be done after the existing PR is merged into master (as it will be very hatd to merge otherwise).

JanGut, 2019-09-19:
This is now resolved with #108.

closed 2019-09-19 by JanGut