bit-bots/bitbots_vision #174: catkin build fails when workspace is python3 and numpy (python2) is not installed

ftsell, 2019-12-10:
When the containing catkin workspace and the used ROS distribution is configured to use python3 and only python3, `catkin build` fails with the error `ImportError: No module named numpy`.

This is due to CMakeLists.txt:56 tries to install vision extensions in python2.

## Expected behavior

The bitbots_vision package should install in both python2 and python3.

## Current behavior

Build fails in a completely python3 environment

## Steps to Reproduce

1. Dont install numpy in python2
2. Build bitbots_vision in python3

## Context (Environment)

- [ ] RViz
- [ ] Simulator
- [ ] Robot
- [x] Local

I did not install numpy for python2. This was done intentionally.

## Possible Solution

There should be a CMake function to get the correct python interpreter. I know CMake uses it internally. If you want to I can look it up later.

JanGut, 2020-10-26:
Is this still a thing?

timonegk, 2020-11-01:
It was closed in 31f5b7dd866a74c3c6ee8c9d35d3b5d8f979bcb0.

closed 2020-11-01 by timonegk