bit-bots/bitbots_vision #180: Field boundary detector: Restep for downsampling detector

Flova, 2019-12-18:
## Is your feature request related to a problem? Please describe.
The downsampling field boundary detector has the tendency to overshoot and leave a gap between the detected and real boundary. This is caused by the bluring.

## Describe the solution you'd like
For the downsampling field boundary detector, we should add some kind of "restep size" to step back and reduce the overshooting of this method.

jaagut, 2021-03-01:
Should this be a constant "restep size" or e.g. depending on the slope of the field boundary in that point?

Flova, 2021-03-01:
No it should depend on the amount of smoothing that was set and that introduces the overstepping.

PR #271 (2021-03-02): Add variable restep for field boundary detector
## Proposed changes
Computes restep using the roi height,

## Related issues
Closes #180
closed 2021-03-02 by Flova