bit-bots/bitbots_vision #246: Dynamic colorspace uses 100% cpu

timonegk, 2020-11-20:
## Expected behavior
The dynamic colorspace should take a reasonable amount of cpu.

## Current behavior
The dynamic colorspace uses a whole cpu.

## Steps to Reproduce
1. `roslaunch bitbots_vision vision_startup.launch`

## Context (Environment)
- [ ] RViz
- [ ] Simulator
- [x] Robot
- [x] Local

Flova, 2020-11-20:
The dyn colorspace has indeed no cap, because the jetson had an unused core for it. We could either add a limit of 1 FPS or something like that or we disable it. I mostly disabled it during testing on the whats your opinion?

SammyRamone, 2020-11-21:
Well, the dynamic colorspace is there for a reason. I would not want to disable it. Just limiting the FPS would be fine and easy todo. We still have enough CPU power available, it's not like we desperatly need to make room. Currently, it just looks like wasting it

PR #248 (2020-11-28): bitbots_vision: Limit dyn colorspace rate
## Proposed changes
This PR ratelimits the dynamic color space node using a new `dynamic_color_space_max_fps` parameter.

## Related issues
Closes #246.

## Necessary checks
- [x] Update package version
- [x] Run `catkin build`
- [x] Write documentation
~- [ ] Create issues for future work~
- [x] Test on your machine
~- [ ] Test on the robot~
- [x] Put the PR on our Project board
closed 2020-11-28 by jaagut