bit-bots/bitbots_vision #300: About The Camera

RoboCary-cpu, 2022-03-30:
I use the Allied Vision camera whose frequency of taking pictures is 60.
The frequency of alternating current in our country is 50Hz. This results in a strobe phenomenon in our industrial cameras. Does this phenomenon occur with the industrial cameras you use? If there is, how do you solve it?

Flova, 2022-03-30:
We just run our shutter at 10-20 Hz because of computational limitations later on in the pipeline, so we never experienced this issue.

RoboCary-cpu, 2022-03-30:
How did you avoid blurry images caused by the robot turning its head?
Another question about lens. Generally speaking, the focal length of industrial cameras cannot be adjusted automatically. How do you choose your lens?

Flova, 2022-03-30:
Due to the relatively large sensor, we are able to reduce the shutter speed so much that we have no motion blur for reasonable motions. We also increase the camera gain, which introduces noise, but enables lower shutter times at the same brightness. There is a tradeoff between the sensor noise and the motion blur. 
Relevant docs: http://doku.bit-bots.de/package/bitbots_vision/latest/manual/tutorials/basler_camera_setting.html

The focal length was chosen by trading of a higher field of view (less head movement, more is observed at once) with a shorter range at which we detect objects (they are smaller in a high fov image). We went for a lens + camera with a field of view of ~45 degrees iirc.. The field of view not only depends on the focal length, but also on the sensor size, so it depends on your camera. We calculated the rough values and rented multiple lenses from https://www.kowa-lenses.com/ that fitted our specifications. These lenses were tested on the robot and a final one was selected and ordered. 

Regarding the fixed focus. We focused our lenses near infinity, as farther away objects need to be the sharpest. We accept a slight blur of objects that are close up as they are very large either way. The f-stop of the camera can make the sharp area larger, but reduces the brightness, which would add motion blur if we increase the shutter. So everything is a tradeoff when working with cameras.

jaagut, 2022-03-30:
Additionally, our camera uses a global shutter instead of the more common rolling shutter to eliminate distortions caused the rolling shutter.

RoboCary-cpu, 2022-04-10:
Thanks a lot,I changed to a better lens and solved part of the problem according to your solution！！！

closed 2022-04-10 by RoboCary-cpu