bit-bots/bitbots_vision #301: Power supply

RoboCary-cpu, 2022-04-10:
Does your camera use the network port to transmit? How do you solve the power supply problem of the camera?

Flova, 2022-04-10:
We power our camera using power over Ethernet. Our "CORE" board handles fuses, motor communication, power management and is also able to add power to a power over Ethernet connection. This approach is quite tightly integrated using custom electronics. We also used a normal 5v voltage regulator to deliver power to the cameras 5v input in the past. For the Basler cameras a properitary power connector was needed. But we switched to the power over Ethernet solution as we can drop the external voltage regulator as well as the additional properitary power cable. Overall it mostly depends on what your cam  supports and how much you want to customise your electronics.

RoboCary-cpu, 2022-04-11:
wow, your power regulation board is so cool!!!!
we also do the power regulation ourselves but the POE is too diffiult to implement. So finally, we decide to use the camera which has a usb3.0 port .

closed 2022-06-13 by jaagut