bit-bots/bitbots_vision #324: Legacy vision dies when detecting couloured obstacles

jaagut, 2023-02-21:


fibonacci35813, 2023-03-05:
Could you explain what exactly is the problem?

jaagut, 2023-03-05:
Hi @fibonacci35813!
Unfortunately, I can not exactly describe the issue, as it was discovered months ago and was only noted down internally. We haven't addressed this issue yet, as we mainly use the newer alternative vision pipeline, including YOEO. With that, the issue does not occur.

In #325, we began resolving the issue.

I am surprised by your question, as this software is fairly tailored to our RoboCup humanoid robot soccer domain. How did you find this issue? How and why do you want to use it? If you are connected to a RoboCup team, please don't hesitate to contact us. We are always open for collaboration.

phinik, 2023-09-19:
Shall we close this issue since we are about to remove the legacy vision anyway?

closed 2023-09-19 by jaagut