bit-bots/bitbots_vision #336: Errors in Coordinate Transformation

qiushuijushi, 2023-05-06:
Hello, I would like to ask a question related to robot vision. When I converted the pixel coordinates of the ball into world coordinates, I found that even a small angle deviation in the camera's pitch can cause significant changes in the ball's position. However, the pitch angle provided by the robot's coordinate system has certain errors (mechanical errors, uneven ground). How do you handle this problem? Thank you.

Flova, 2023-05-07:
These errors are very typical when it comes to inverse perspective mapping. There is only so much you can do about them (after a few meters it gets unreliable even if you tune your system very good). The errors increase the closer you get to the horizon as a smaller angular difference corresponds to a larger distance in world coordinates. Here are a few best practices to minimize errors during ipm:

- Measure the rotation (pitch and roll) of your camera with respect to the ground coordinate system. Translation is pretty much irrelevant. You can do this by using a IMU, preferably in the head. We merge our IMU data into the robots tf tree using the odometry fuser node.
- Get the timings right. The robot moves quite a bit during walking, therefore dynamics are also important during inverse perspective mapping. Use a fast shutter speed and preferably a global shutter to capture your image. I recommend using an industrial grade camera for this. Estimate transition latencies and add a very accurate time stamp to the camera image message. This time stamp will be propergared trough the pipeline. The detected balls etc. will also be equipped with the timestamp of the original image. Never use the current time, as the robot moves quite a bit during the processing. When you perform the ipm into world coordinates, use the time stamp of the object (e.g a ball) to query the IMU and joint states (forward kinematics) from the tf buffer at the time when the image was taken. 
- Perform intrinsic calibration. Calibrate the robots camera using a checkerboard to estimate the intrinsics (distortion, focal length etc.). Crosscheck these values roughly with ideal ones calculated from camera/lenses datasheet. Also perform a rectification of the image if you use a wide angle lens, use the distortion values estimated during the calibration.
- Perform intrinsic calibration. Small constant offsets in the orientation of both the IMU and camera result in ipm errors. Tune per robot rotation values for both of them. You can do this manually by dynamically adjusting the robots tf tree until e.g. the projected field lines match a predefined map. Make sure a proper intrinsic calibration is performed before this.

This should get you quite far in terms of ipm accuracy. For long distances errors will be present ether way, as subdegree orientation differences have large impacts. But the accurate position estimation of far away objects is also less important in RoboCup as very precise data is mostly needed when the robot is close to the ball.

Flova, 2023-05-07:
I can also recommend you a standardized suite of inverse perspective mapping packages for ROS 2:
https://github.com/ros-sports/ipm

There is also a RoboCup specific package: https://github.com/ros-sports/soccer_ipm

Flova, 2023-05-07:
There are also other methods for depth estimation in RoboCup. They all have their benefits and drawbacks. We wrote for example a paper on monocular depth estimation in the RoboCup context: https://www.researchgate.net/publication/366192714_Applying_Monocular_Depth_Estimation_in_RoboCup_Soccer

closed 2023-08-17 by jaagut